amymcgovern/pyparrot #209: Having an issue with WifiConnection.py

Brammerz45, 2020-11-24:
I am using python 3.9 and I am new to pyparrot. When trying to connect to my Bebop 2 I come across the following error:

  File "C:\Users\Michael\AppData\Local\Programs\Python\Python39\lib\site-packages\pyparrot\networking\wifiConnection.py", line 314, in _handshake
    self.drone_ip = ipaddress.IPv4Address(self.connection_info.address).exploded
AttributeError: 'ServiceInfo' object has no attribute 'address'

The code bringing up the issue is 
[image]

I have looked up which attributes should replace `self.connection_info.address` and tried `self.connection_info.addresses`, `self.connection_info.addresses_by_version()`, `self.connection_info.parsed_addresses()`
and each of those bring different errors. 

Any help to resolving this issue would be great, thanks.

Ren3t, 2021-01-22:
I have the same issue with Bebop 2...

Brammerz45, 2021-02-01:
For anyone else who gets this error, it is solved by reverting to an earlier version of zeroconf. This is done easily in pip via the command `pip install zeroconf==0.20.0` Now I reverted to 0.20.0 and it worked instantly but I dont know what the most up to date version of zeroconf this will work with.

roverico, 2021-03-25:
Hello,
In my case we have solved this issue using this sentence against the Sphinx simulator (within a WMWare VirtualBox using WiFi dongle compatible Tp.link)
 self.bebop = Bebop(drone_type="Bebop2", ip_address="10.202.0.1")
and
 self.bebop = Bebop(drone_type="Bebop2", ip_address="192.168.42.1") against the real bebop2 (using Administrator permissions and disconnecting any firewall/antivirus/spyware)

Cheers,
roverico

jannacek, 2021-06-08:
I solved the issue by downgrading to zeroconf version 0.20.0, but now I get a new problem:

ConnectionRefusedError: [Errno 61] Connection refused 
:( 

any help please? :)

muelly2, 2021-10-03:
There was a change in zeroconf, where you have to change all instances of 
`self.connection_info.address`
to
`self.connection_info.addresses[0]`
within _wifiConnection.py_

theodom, 2022-04-07:
> There was a change in zeroconf, where you have to change all instances of
> `self.connection_info.address`
> to
> `self.connection_info.addresses[0]`
> within _wifiConnection.py_

I did this exact thing, I changed all the lines that contains `self.connection_info.address' but it still gives me basically the same error: 
```print (self.connection_info.addresses[0], self.connection_info.port)
AttributeError: 'ServiceInfo' object has no attribute 'addresses'```
I have zeroconf on version 0.20.0.