amymcgovern/pyparrot #223: is it possible to connect parrot mambo to a wifi network ?

Gseeker-My, 2023-04-26:
Hello !
am an avionics student , and i am trying to use parrot mambo for object detection ,my approach is to get the rtsp stream from the wifi camera and process the images on the laptop then control the mambo using wifi also ,the problem im facing is that im using other wifi cameras too for depth estimation , and i need all the devices in one network , so is it possible to make the mambo drone connect to a wifi network rather then sharing his own network

amymcgovern, 2023-04-26:
I am no longer maintaining this as parrot stopped producing them and my drones needed replacement.  However, this is a firmware issue from parrot and was not possible as of the last time I had it running.  Maybe they have updated but I doubt it.

Gseeker-My, 2023-04-26:
i see , thank you for your comeback